OpenFTC/OpenRC_2017-18 #10: Release to Play Store?

NoahAndrews, 2017-10-25:
If we're going to release to the Play Store (and I think we should), we need to consider how that will work.

* The identifier will be different
* We'll need to step out of sync with the official app's versionCode, so that we can issue updates outside of their release cycle
* Which identifier should the website's downloadable APK have?
*

FROGbots-4634, 2017-10-30:
Per our discussion, we will release the Blockly & OBJ Sync app as an entirely separate application.

closed 2017-10-30 by FROGbots-4634

magneticflux-, 2017-10-30:
I'm not sure how FIRST gets around these regulations, but per the [Play Store policies](https://play.google.com/about/privacy-security-deception/malicious-behavior/):
>An app distributed via Google Play may not modify, replace, or update itself using any method other than Google Play’s update mechanism. Likewise, an app may not download executable code (e.g. dex, JAR, .so files) from a source other than Google Play.

>The following are explicitly prohibited:
>...
>Apps or SDKs that download executable code, such as dex files or native code, from a source other than Google Play.

Just something to watch out for. FIRST might have lawyers or connections to Google, but a community project wouldn't really have anywhere to go if it were removed from the Play Store.

FROGbots-4634, 2017-10-30:
@magneticflux- You're thinking that OnBotJava and Blockly are violations?

NoahAndrews, 2017-10-30:
Seems like a little bit of a gray area to me, but probably fine. The FTC app is not downloading executable code from anywhere. But, it does allows users to write programs that can be run from within the app. It's an interesting question whether or not that violates that policy.

At any rate, we're not planning on publishing our version to the play store anymore.

magneticflux-, 2017-10-30:
@FROGbots-4634 If OnBot Java truly ran on the device, it would be clear-cut IMO. However, the `.java` files are sent over the network (hopefully encrypted?) from a user's computer. That is almost certainly "downloading from a source other than Google Play". I know that OnBot Java uses a micro webserver to host the IDE, but does Blocky exclusively run on the device?

FROGbots-4634, 2017-10-30:
Well, do consider that it's not downloading any prewritten code. The user would have to actively write the code and then download it.

But again, it's a moot point since we're not going to be releasing to the Play Store.

Also if note would be the fact that the person who heads up blocks and OBJ development is a Googler, so I'm sure she take care of any legal issues.

magneticflux-, 2017-10-30:
@FROGbots-4634 That's true, it says nothing about downloading sources and compiling them on the device. It's probably for the best that OpenFTC won't be on the Play Store, since there's no real point to not having the source code of it and the "turbo" version is totally useless without stuff in TeamCode.

NoahAndrews, 2017-10-30:
Yeah, the only reason I was going to put it on the Play Store was because I was considering enhancements for Blocks and OnBotJava users. If I or someone else comes up with a nice enhancement for them, I may revisit this.