OpenFTC/OpenRC_2017-18 #3: Contribution Guide

NoahAndrews, 2017-10-25:
* Pull requests must be made from a dedicated branch. Do not submit a pull request for your forked master branch.

FROGbots-4634, 2017-10-25:
Any of FIRST's code which you have commented out, modified, or removed should be notes with `//modified for Turbo`

WardBenjamin, 2017-10-25:
Just as a note, following standard Java and Android comment conventions, that should be:

`// Modified for OpenFTC`

or

`// Modified for Turbo`

with a leading space and capitalized first letter.

NoahAndrews, 2017-10-25:
Good point.

NoahAndrews, 2017-12-14:
Any code that would require a dependency on another module should be kept out of the OpenFTC module, and likely placed in the same module as the code it depends on. This is to prevent circular dependencies. 

If possible, place new classes in other modules in a `org.openftc.robotcore` package (for example)

NoahAndrews, 2017-12-20:
[image]

NoahAndrews, 2017-12-21:
You can help by giving yourself the tester role on Discord