OpenFTC/OpenRC_2017-18 #5: Set up a Continuous Integration server

NoahAndrews, 2017-10-25:
Key functionality
- Verify that all PRs build successfully
- Report the output APK size
- Ideally, report app installation time on a Moto G4 Play

I believe reporting that information at the pull request might require a bot. For an initial version, it's OK if we have to go look at the CI output.

magneticflux-, 2017-10-27:
I set up Travis CI for Pattonville Robotics's shared team library [here](https://github.com/Pattonville-Robotics/Common-Code/). We don't have APK size reporting, but it verifies that PRs build and it runs any tests that may exist. APK size can also get included in the Travis CI log by just printing it out after the artifact is built.

NoahAndrews, 2017-12-14:
This should assemble and publish the javadocs