OpenFTC/OpenRC_2017-18 #6: Ease OpenCV development

NoahAndrews, 2017-10-25:
* I like the idea of having a button somewhere that will download the OpenCV lib to the app's storage. See @FROGbots-4634 if you're thinking about working on this.
* Add comments with additional pain points, libraries you think we should include, any enhancements related to OpenCV.

I don't have any experience here, so this will have to be contributed by someone else.

WardBenjamin, 2017-10-25:
Realistically, the only way to add OpenCV support is by modifying the FtcRobotControllerActivity. I've pushed a copy with OpenCV support to the `opencv` branch of [my fork](https://github.com/WardBenjamin/OpenFTC-app) but I have a few issues:

- [ ] Requires every team modify the FtcRobotControllerActivity
- [ ] Difficult to understand
- [ ] Difficult to modify
- [ ] Requires bundling the entire repository of compiled OpenCV binaries (This could be reduced by only supporting the platforms for the phones specified in GM1)

And I'm sure there's a whole other list of issues as well. As of right now, the setup is very similar to that done by LASA Robotics and the electronVolts but I think it could be improved a lot.

Thoughts?

FROGbots-4634, 2017-10-25:
I don't see why it needs to go in the FtcRobotController activity.... why can't it be a separate class used from an OpMode?

WardBenjamin, 2017-10-25:
@FROGbots-4634 Most of the code can, but if we want to view the camera feed as per Vuforia you need to hook into the activity layout. There _is_ a way to do that dynamically but it still requires hooks into the activity.

nicholasday, 2017-10-29:
I've hooked into it dynamically without modifying the FtcRobotController activity. 

You can write to a surfaceview that is added to the R.cameraMonitorViewId during init of an OpMode:

```java
        context = hardwareMap.appContext;
        final Activity activity = (Activity) hardwareMap.appContext;

        activity.runOnUiThread(new Runnable() {
            @Override
            public void run() {
                LinearLayout l = (LinearLayout) activity.findViewById(R.id.cameraMonitorViewId);
                surfaceView = new MySurfaceView(context, WIDTH, HEIGHT);
                l.addView(surfaceView);
            }
        });
```

FROGbots-4634, 2019-08-21:
[EasyOpenCV](https://github.com/OpenFTC/EasyOpenCV) is now a thing

closed 2019-08-21 by FROGbots-4634