OpenFTC/OpenRC_2017-18 #8: Warn the user about other FTC apps installed on the phone

NoahAndrews, 2017-10-25:


WardBenjamin, 2017-10-31:
Just as a note, there's some code to detect if the DS is installed in the Inspection Activity. From there, it should be as easy as follows:

```
final String appPackageName = getPackageName(); // getPackageName() from Context or Activity object
try {
    startActivity(new Intent(Intent.ACTION_VIEW, Uri.parse("market://details?id=" + appPackageName)));
} catch (android.content.ActivityNotFoundException anfe) {
    startActivity(new Intent(Intent.ACTION_VIEW, Uri.parse("https://play.google.com/store/apps/details?id=" + appPackageName)));
}
```

FROGbots-4634, 2017-10-31:
Thanks; I actually already wrote the code for doing that yesterday, just haven't PRed yet