OpenFTC/OpenRC_2017-18 #9: Come up with a good versioning system

NoahAndrews, 2017-10-25:
The turbo repo kind of has a tacked-on "turbo version". We need to integrate that with the main repo somehow, decide on if/how we want to express that in versionName, etc. If we're publishing to Google Play, that will affect this.

FROGbots-4634, 2017-10-25:
Yeah I think syncing the version codes would be too much trouble

NoahAndrews, 2017-10-25:
It wouldn't be any trouble. We just never touch versionCode. It only gets updated when FTC updates it.

NoahAndrews, 2017-10-31:
Definitely switch to a proper BuildConfig for the OpenFTC module. We want RobotLog.logBuildConfig() to be safe to use.

WardBenjamin, 2017-10-31:
As long as the three required variables are present, the OpenFTC class is logged identically to any BuildConfig class, since RobotLog just pulls the three named static variables.

WardBenjamin, 2017-11-01:
Because of our dependency on the official ftc_app version, I would suggest following the Electron versioning system (a variant of SemVer due to Electron needing to follow version updates of it's Chromium and Node dependencies):

# Electron Versioning

Due to its dependency on Node and Chromium, it is not possible for the Electron project to adhere to a SemVer policy. You should therefore always reference a specific version of Electron.

## Electron version numbers are bumped using the following rules:

**Major** is for breaking changes in Electron’s API. If you upgrade from 0.37.0 to 1.0.0, you will have to make changes to your app.
**Minor** is for major Chrome and minor Node upgrades, or significant Electron changes. If you upgrade from 1.5.0 to 1.6.0, your app is supposed to still work, but you might have to work around small changes.
**Patch** is for new features and bug fixes. If you upgrade from 1.6.2 to 1.6.3, your app will continue to work as-is.

Source: https://electron.atom.io/docs/tutorial/electron-versioning/

NoahAndrews, 2017-11-04:
For OpenFTC-turbo, I'm not sure it makes sense to have a separate "turbo version". I think that's kind of confusing, especially when the way things are set up results in a variable called `VERSION_COMPLETE` with a value of `OpenFTC-turbo 1.0`. Is that 1.0 for the Turbo Version or 1.0 for the OpenFTC version? 

My proposal is that we get rid of the "Turbo version", and keep the version numbers the same between them. In the rare case where we want to make a new release only for the Turbo version, we simply append `-1` to the end of the OpenFTC version number. @FROGbots-4634, the "Turbo version" was your idea, so I'd like to hear your thoughts especially.

FROGbots-4634, 2017-11-04:
Well I feel like the Turbo version is good because we may need to be removing more and more stuff as time goes on. For example, I've been meaning to nuke the 2016-17 Vuforia target files but I haven't PRed yet.

NoahAndrews, 2017-11-08:
Yeah, but it's confusing to see "OpenFTC-turbo 1.5" (made-up version numbers here) on the main screen and then "Turbo version: 1.1" on the about screen. I don't see why we need a separate number to track it. If a user is running OpenFTC-turbo version 1.5, we already know what changes to the "turbo" part that includes, and so will the user if they look at the release notes.

NoahAndrews, 2017-12-01:
I think we should say that "minor" version bumps indicate either a new version of the official SDK has been included, OR a new OpenFTC feature was added that changes something big.

For example, DSktop support would be a "patch" bump, and the ability to build a stock APK from an OpenFTC project would be a "minor" bump.

closed 2018-02-09 by NoahAndrews