OpenFTC/OpenRC_2017-18 #11: Utilize Maven repository for versioning

magneticflux-, 2017-10-27:
I created and am maintaining a public Maven repository that hosts `.aar`s from the `ftc_app` repository. Currently, Pattonville Robotics (2866, 2867, 7856) use this repository to manage transitive dependencies with a our shared library "Common Code". Here is the repository: [link](https://github.com/Pattonville-Robotics/ftc-lib-repo).

Would you be interested in using this for your build scripts?

NoahAndrews, 2017-10-29:
I can't say I really see an advantage to setting things up that way. We can download the files from GitHub easily enough, and we're not dependent on a third party that way.

Thanks for the offer though! Your common code library looks very cool. Is it intended to be open source?

closed 2017-10-29 by NoahAndrews

magneticflux-, 2017-10-29:
@NoahAndrews Right now it doesn't have a stable public API, so we're cautious about releasing it. We should definitely license it eventually though. I'm thinking LGPLv3 or something similar.