OpenFTC/OpenRC_2017-18 #15: Repository size concerns

magneticflux-, 2017-10-29:
Since merging from `ftc_app` is going to be impossible due to the structure changes, it doesn't make sense to preserve the history from before the structure changes. If completely wiping the history should be avoided, I have had success using the `git-bfg` tool to strip large blobs from Pattonville Robotics's repositories. Our net change was around 220MB -> 19MB on average.

The earlier this is done the better because it requires a redownload for everyone with old copies of the repository history.

I spent the time waiting for my fork to clone writing this issue lol. And the results are in! A whopping 522MB for this repository! Something should probably be done about that...

magneticflux-, 2017-10-29:
The storage of many binaries in this repository is also cause for concern. Their size will only increase with each update. Git LFS is the "correct" solution to large binaries, but that could be eliminated using Maven coordinates. If portability is still desired, you can add a local Maven repository to the buildscript and install binaries there with `mvn install:install-file`.

FROGbots-4634, 2017-10-30:
*"Since merging from ftc_app is going to be impossible due to the structure changes"*

But we're expressly trying to maintain the ability to merge from ftc_app. We will avoid making any very large structural changes

WardBenjamin, 2017-10-31:
I agree. In my mind, this isn't just a Turbo feature - the official repository recommends cloning with a depth of 1 for this same reason. Since we don't need to preserve all of the history, as this code isn't designed to be merged back into ftctechnh/ftc_app, I think stripping the large blobs would only help us.

NoahAndrews, 2017-11-04:
I'll experiment with this when I get a chance, but we have users, and I don't see us modifying the public history. Sorry. :/ We'll definitely add a recommendation for a shallow clone at least.

WardBenjamin, 2017-11-07:
As you might know, FIRST currently recommends that teams clone with the clone depth `--depth` set to a low number (1). Using the same idea, we can set up OpenFTC-app as a "shallow remote"
```
; Shallow to last 5 commits
git rev-parse HEAD~5 > .git/shallow

; Manually remove all other branches, tags and remotes that refers to old commits

; Prune unreachable objects
git fsck --unreachable ; Will show you the list of what will be deleted
git gc --prune=now     ; Will actually delete your data```

This example trims all but the last 5 commits, and keeps identical commit hashes. This allows the ability to push/pull from both OpenFTC-app and the official ftc_app since they have "identical" histories. People using Git <1.9 won't be able to pull from a shallow remote if they have the complete repo.

(To be continued, I have to head out.)

NoahAndrews, 2017-11-07:
That sounds incredibly promising, thanks Ben!

NoahAndrews, 2017-11-07:
Unfortunately, to prevent the size from growing again, I think I would have to stop using git to pull down new versions from FIRST, which is how my script currently works.

WardBenjamin, 2017-11-07:
@NoahAndrews I'll have to test, but I believe you should be able to pull from the official version and still keep a shallow copy. 

The downside (and upside? Call it a limitation.) of using the shallow copy like this is that we're limited to only the history that we choose - meaning that we won't preserve history from past seasons or whatnot. This _should_ allow for direct push/pull/merge since it doesn't rebuild commits, so the repository state should look "the same" to git (as new commits are just diffs based off of the last one.)

The code that I posted above is functionally the same as:

```
; Clone the original repo into limitedRepo
git clone https://web.site/originalRepo limitedRepo --depth=5

; Overwrite remote with limitedRepo
git push origin master --force
```

NoahAndrews, 2017-11-10:
@WardBenjamin does this whole thing provide any benefit over a user doing a shallow clone themselves? (aside from the obvious one that they don't have to remember to do it)

NoahAndrews, 2017-12-14:
We're not going to do this. No rewriting history. Teams should perform a shallow clone.

closed 2017-12-14 by NoahAndrews