OpenFTC/OpenRC_2017-18 #17: Put each module in its own repository

magneticflux-, 2017-10-30:
If both ease of use *and* ease of modification are priorities, I feel that having each module (that has sources, at least) occupy its own repository is a step in the right direction.

To address ease of use, [JitPack](https://jitpack.io) is available. You can find versions of Pattonville Robotics's Common Code hosted and versioned using JitPack [here](https://jitpack.io/#Pattonville-Robotics/Common-Code). End users would merely have to include JitPack as a Maven repository and then reference each library by their coordinates.

To address ease of modification, [Composite Builds](https://docs.gradle.org/current/userguide/composite_builds.html) are available. This feature allows users to clone, for example, the `RobotCore` dependency's sources and then replace all uses of the 'official' `RobotCore` artifact with the results of building the current `RobotCore` sources. Android Studio 3.0 is advanced enough to recognize Composite Builds and seamlessly integrate them into a project, and so far this process has worked very well in our Common Code project for immediately testing changes and then marking releases on `master`.

This is an example of how Composite Builds may be used, taken from our [current testbed repository](https://github.com/Pattonville-Robotics/2866/blob/2f98e4d62fd8e3b3227852539638f908cd06c9b8/settings.gradle):

```Gradle
def toggleCommonCode = false
gradle.ext.include_common_code = toggleCommonCode && file('../Common-Code').exists()
if (gradle.ext.include_common_code) {
    includeBuild('../Common-Code') {
        dependencySubstitution {
            substitute module('com.github.Pattonville-Robotics:Common-Code') with project(':library')
        }
    }
}
```

FROGbots-4634, 2017-10-30:
I'm not sure I really understand how it would improve ease of use/modification... (You didn't give any examples)

Also, one very good reason against doing this is that some changes will require modifying files in multiple modules

NoahAndrews, 2017-10-30:
How would this improve ease of use and ease of modification? It seems to me that this would make modification _more_ difficult by requiring that users make changes to the Gradle files if they want the ability to even casually tweak stuff.

magneticflux-, 2017-10-30:
@FROGbots-4634 I believe that it would improve ease-of-use for all teams except those that plan to completely overhaul the internals (of which there are not many). By separating the modules, teams can choose to modify (by forking and referencing their fork though JitPack) individual modules without having to manage the entire project.

When it comes time to update from, say, v3.4 of the ftc_app to v3.5, all the teams would have to do for unmodified modules is change a version number and for modified modules is merge their upstream and fix any conflicts. Compare that to having a monolithic repository where updating means trawling through a massive merge and making sure none of your modifications to internals are overridden or incompatible. Having separate repositories would also greatly reduce the download size of the main repository, so teams would only have to download what they absolutely need.

Additionally, compiling modules that aren't going to be modified slows development time greatly. While Gradle's build cache can help with that to some extent, even unmodified modules must be recompiled and redexed under some circumstances. Having `.aar`s available for typical use will speed up build times.

Finally, specifying transitive dependencies in a `pom.xml` using the `android-maven-gradle-plugin` lets Gradle avoid duplicating dependencies. Right now, including `com.android.support:support-annotations:27.0.0` in a TeamCode build causes a Dex error because the annotations are already partially included in RobotCore. Telling Gradle about these transitive dependencies lets its dependency resolution system automatically pick the newer version or error and request clarification, rather than proceeding and eventually failing when dexing.

NoahAndrews, 2017-10-30:
> By separating the modules, teams can choose to modify (by forking and referencing their fork though JitPack) individual modules without having to manage the entire project.
>
> When it comes time to [...]

I don't see what there is to "manage". If you don't touch the code in a module, you won't have any merge conflicts there. Unless...

Are you thinking that to update to a new version, you'd pull from the official ftc_app repository? Because yeah, you're right, that would be terrible. Not even just because it might conflict with our changes to the core modules, but because the updated core modules would be provided in binary form. To merge the two, you'd need to extract the aars and source jars over what's there, which is a nightmare to do to code that you've modified. 

My plan for updates is to always update to new SDK versions within 5 days of their release, assuming FIRST doesn't majorly refactor the app in that particular release. I'll worry about any conflicts with new FIRST code (the `unmodified` branch gives me a place to do the extraction without worrying about overwriting stuff), you just have to pull the new version from OpenFTC. There will only be merge conflicts with files that the user modifies. And that would be true regardless of whether the modules were in separate repositories or not. This will all be fully documented as soon as possible.

> Having separate repositories would also greatly reduce the download size of the main repository, so teams would only have to download what they absolutely need.

I don't see how that's true. Teams have to download all of the modules one way or another. Even if you decided to remove the Blocks module (the only open one that isn't absolutely necessary) you'll run into a lot of errors that have already been dealt with in the Turbo version.

> Additionally, compiling modules that aren't going to be modified slows development time greatly. While Gradle's build cache can help with that to some extent, even unmodified modules must be [...]

Can you clarify what "some circumstances" would be?

I need to do some more investigation into the performance and transitive dependency issues before I comment on them, I think. But if for no reason other than the one @FROGbots-4634 brought up, I don't think this is a viable solution.

closed 2017-10-30 by NoahAndrews

NoahAndrews, 2017-10-30:
I opened a new issue (#17) for the performance and transitive dependency issues. Anything you want to say regarding that specifically should go there.