OpenFTC/OpenRC_2017-18 #39: Error: Robot failed to start: Attempt to invoke.... ConfigurationDevice.isDeviceFlavor() on a null object reference

WardBenjamin, 2017-12-08:
Full error:
[image]

See RC log: https://gist.github.com/WardBenjamin/112e8b43ed487c52b4d96da8ac2471f7
See config: https://gist.github.com/WardBenjamin/77420d2f75d2cd2479ce642f69574535

NoahAndrews, 2017-12-08:
Also of note: the config screen shows no config files, and scans turn up empty.

WardBenjamin, 2017-12-08:
RC log for commit hash e5003f776024a7bf0bd16ef80e0cba87c5309a22: https://gist.github.com/WardBenjamin/fe73751eeb12a5756a8366d791de2240

NoahAndrews, 2017-12-08:
Access to the `mapTagToUserType` object is not thread-safe. Worth noting.

NoahAndrews, 2017-12-08:
Might be a 3.5 issue, it only showed up after updating.

NoahAndrews, 2017-12-08:
I'm expecting to see a "Filtering UnspecifiedMotor class" message.

NoahAndrews, 2017-12-08:
What the log should look like (with latest logging additions): https://gist.github.com/NoahAndrews/dd331687e2b9b8a460b0d8b6a8ba23fb

FROGbots-4634, 2017-12-08:
@WardBenjamin Have you tried:

 - Build -- > Clean
 - Build --> Rebuild
 - File --> Invalidate Caches / Restart
 - Completely uninstalling the RC app before deploying from AS

(I sometimes have an issue where the RC says I have duplicate OpModes even though I don't and one of the above steps fixes it)

NoahAndrews, 2017-12-08:
@WardBenjamin As a reminder, please run my latest logging code before you try to fix the issue by any of those steps.

WardBenjamin, 2017-12-09:
RC log for commit hash e3e6643408285a5c74779d28cfa9ba574fe7c3c8:
https://gist.github.com/WardBenjamin/68612d108e223573a7b3d9e4109093a6

NoahAndrews, 2017-12-12:
This turned out to be an Instant Run issue. Which I think I've actually fixed.

closed 2017-12-12 by NoahAndrews