OpenFTC/OpenRC_2017-18 #46: Dynamically load OBJ and Blocks like the Vuforia mod?

FROGbots-4634, 2017-12-18:
Would it be possible for us to package all the stuff that got stripped out for Turbo into an AAR which could then get dynamically loaded from the SDcard like my Vuforia mod?

NoahAndrews, 2017-12-18:
So, one objection I came up with for this is that it prevents OBJ and Blocks modification

NoahAndrews, 2017-12-18:
And if we can get that to work, maybe there's other stuff we should do this with too

NoahAndrews, 2017-12-18:
And if we can't, we can at least load android.jar from SD

NoahAndrews, 2017-12-19:
[image]
[image]
[image]

NoahAndrews, 2018-01-08:
OnBotJavaManager actually copies the contents of `Robotcore\assets\java` to `FIRST\libs` already. Which would make it VERY easy to not include that stuff in the APK.

Javac.jar is already isolated, so it shouldn't be too hard to load that at runtime too. I think the biggest savings after that would be to put the `org.firstinpires.ftc.robotcore.internal.android` package into its own jar. At that point, we can see how far we are size and method wise from Turbo. Of course, even if we decide that it's slim enough, we still should measure the impact of an option to disable the HTTP server and stuff.