OpenFTC/OpenRC_2017-18 #50: Duplicate OpMode Name error

kskrueger, 2017-12-21:
As an example, when duplicating an OpMode, occasionally you forget to rename the name string defined as @Teleop(name="Name Here", group="Test"). If you happen to forget to change the name, there is no warning until it is already loaded on the phone, in which the app will not let you run ANY OpModes at all. Is it possible to have this error not block all OpModes from being run, or have a script that checks and warns you before downloading possible? Not too big of a problem...just an idea...

rbrott, 2017-12-21:
Seems possible with annotation processing during compilation.
https://docs.oracle.com/javase/8/docs/api/javax/annotation/processing/AbstractProcessor.html
https://developer.android.com/studio/build/gradle-plugin-3-0-0-migration.html#annotationProcessor_config

NoahAndrews, 2017-12-21:
@rbrott Yep, that's the direction I was thinking. Want to take a stab at it?

rbrott, 2017-12-21:
Sure I'll give it a go.

NoahAndrews, 2017-12-21:
Awesome!