OpenFTC/OpenRC_2017-18 #78: Logo for OpenFTC and OpenRC

ssnjrthegr8, 2018-08-08:
So as said on discord the logo is ready and all that is left is to give you the files :) Below are few colour variations of the logo (If OpenFTC has an official colour scheme do tell me and I will implement it) :

[image]

Also I need to know what platforms the app is going to run on.

NoahAndrews, 2018-08-08:
I'll ask the community to give their opinions on this GH issue. I think a bunch of these look great!

The app runs on Android 4.4+ (no devices that currently run Oreo are legal for competition use at this time, but that will change eventually). Also, the app will be forced to target Oreo by November, which I think will require that we add an [adaptive icon](https://developer.android.com/guide/practices/ui_guidelines/icon_design_adaptive). 

iOS is not something that we'll _ever_ need to worry about.

FROGbots-4634, 2018-08-08:
Thinking about color scheme, what if we implemented the FIRST logo color scheme? (red, white, and blue) Just a passing thought....

NoahAndrews, 2018-08-08:
OTOH though, FTC is orange

KPetrucci, 2018-08-08:
One of the ones in the middle two rows might be cool if you put the "o" in the semicircle cut out of the gear, if you can see what I mean. Also I like the orange since it's bright and will stand out from all the gray on a robot, and like NoahAndrews said it's the FTC color.

multiplemonomials, 2018-08-09:
I vote for the bottom right one personally.  It's the inverse of the blue and white icon that the official app uses (used?), plus it just looks cooler.

ssnjrthegr8, 2018-08-28:
Some variations of the original logo:
[image]

nathanlesser, 2019-08-21:
Oh hey finally something I can competently speak about here!

It's good, but the new color scheme just isn't as good as the original. Additionally, the text is pretty small, and the core logo feels more like a settings app than it does an icon for something FTC related.

Way better than the original tho, and nice work!