OpenFTC/OpenRC_2017-18 #79: Update for SDK v4?

qwertychouskie, 2018-09-14:
Any info on when this will be updated for version 4 of the SDK?

NoahAndrews, 2018-09-15:
When I get some time to work on it. Also, there was a MASSIVE amount of change in 4.0. My update script will need some tweaking, and there will be at least a couple of painful merge conflicts to resolve. For example, we modified the About screen to add an About OpenRC button. The About screen implementation got completely replaced in 4.0. 

Hopefully it won't be too long, but I'm not guaranteeing a timeline right now.

NoahAndrews, 2018-09-15:
Note to self: Make sure that we replace the old version of Vuforia stored on the phone

FROGbots-4634, 2018-10-22:
@qwertychouskie we're all busy so we haven't hide time to work on it yet :(

qwertychouskie, 2019-05-10:
So... Still no 4.x?

rbrott, 2019-05-10:
It looks like there will never be 4.x support. According to @FROGbots-4634:
> v4.x is dead in the water, however, v5.x is on the roadmap.

FROGbots-4634, 2019-08-21:
@qwertychouskie [OpenRC-Turbo](https://github.com/OpenFTC/OpenRC-Turbo) has been refreshed and is now based off of SDK v5.0.

closed 2019-08-21 by FROGbots-4634